mikeferguson/robot_calibration #17: need to parameterize checkerboard starting offset

mikeferguson, 2015-06-28:
Currently the checkerboard "world" poses are poorly defined and not easily changed. The orientation is hard-coded for the Fetch configuration as to where the checkerboard points are relative to the kinematic chain tip link. Additionally, there is currently no way to define a starting position, so a number of cycles of the optimizer are spent simply moving the checkerboard to a relatively good starting position.

mikeferguson, 2018-06-06:
Ok, I had to refresh my memory on the workflow of this one as well:

 * First, [in the CheckerboardFinder](https://github.com/mikeferguson/robot_calibration/blob/e3d509ca487bc6d847b00be7b0f4c3ce9f50c5fb/robot_calibration/src/finders/checkerboard_finder.cpp#L203), we fill in the "world" and "rgbd" samples. The rgbd one is fine (that's the coordinates of the checkerboard points within the camera frame).
 * The world one is the one we are concerned with. [It's been hacked in the past for particular robots](https://github.com/mikeferguson/robot_calibration/commit/0cc3420f8c9f4bdc9486e1a3eaa10560cb3eb020#diff-c492b4ea3806f113851c9d45e760b2c7).
 * When we enter the calibration phase, we use a "magic" extra transform (it's not part of any of the KDL stuff, since there isn't a "checkerboard" frame in the robot URDF). [I just added some comments to the key part](https://github.com/mikeferguson/robot_calibration/commit/2ff79fa4f486d1b3c8ced156243029f8fc236423) (since I couldn't actually find it easily with a search for "checkerboard").

So then, during optimization, we are optimizing a "wrist_roll_link"->"checkerboard" offset along with all the other optimization parameters. Ideally, this offset would be almost correct at the start of optimization, especially in angular orientation (otherwise it is the primary source of error). My "hack" above is actually just fixing the orientation part, the linear translation still has to be dealt with by the solver -- and obviously it only works for a particular robot.

In my mind, there are two approaches to fix this:

1. continue on the current path and add more parameters to the checkerboard finder -- this has downsides in that you can't update those parameters AFTER a capture, which makes it challenging to actually do the initial tuning of your parameters as well.
2. revert the "hack" above so that there are no crazy inversions of the coordinate frames, and add the ability to load initial frame offsets during the optimization. This would allow you to specify the initial state of the "wrist_roll_link"->"checkerboard" offset. In theory, it also allows lots of other functionality.

To do approach 2, we need to:
 * Add a new parameter set, perhaps "free_frames_initial_value", to the OptimizationParams. The loader can probably [be modeled off this](https://github.com/mikeferguson/robot_calibration/blob/melodic-devel/robot_calibration/src/optimization_params.cpp#L47).
 * Then those [initial values need to be applied here](https://github.com/mikeferguson/robot_calibration/blob/melodic-devel/robot_calibration/src/optimizer.cpp#L109).

Side note: the orientation part is especially important -- many times you know/desire that the orientation of the checkboard is fixed in a plane defined by the gripper (the fingers hold the checkerboard parallel to the fingers -- so there is only rotation about one axis). For instance, with Fetch, [only pitch was a free param](https://github.com/fetchrobotics/fetch_ros/blob/indigo-devel/fetch_calibration/config/calibrate.yaml#L41) (with the hack above).

Let me know if you're actively working on this -- if not, I can whip up a PR if you're willing to test. I don't currently have access to a robot...

guilhermelawless, 2018-06-06:
I appreciate the huge effort that went into that reply. You have to admit the hack is a bit all over the place.

I don't have enough knowledge of the code to be able to fix it, but I would gladly test it on our robot. Perhaps there could be a way to publish the "world" points along with the "rgbd" points so I could compare them?

Now I am wondering if this is why we're having some issues calibrating. We have other things not working so well (mostly a bad motor in the neck), so I'm not 100% sure.

For the checkerboard I created a virtual link, with X pointing away from the gripper. I think this is similar to fetch. I also am allowing x,y,z and yaw (not pitch) as free params for the checkerboard frame. See image.

[image]

However i'm getting pi/2 offset on yaw from the calibration procedure. This suggests that the hardcoded frame does not expect a similar "virtual checkerboard" frame to ours?

mikeferguson, 2018-06-06:
Ok, branch here with initial offsets: https://github.com/mikeferguson/robot_calibration/commits/checkerboards_that_work

I also removed that checkerboard finder hack -- as that will make it way easier to figure out the correct orientation.

I know you mentioned a neck motor problem -- is it possible to collect another bag file with the checkboard finder hack removed? If you can share that bag file along with your calibration configuration, I could take a look at it.

Two other checks:
 * What camera are you using?
 * Are you sure that the checkerboard scale is correct?

mikeferguson, 2018-06-06:
Also, to get an initial estimate of the orientation, you can do the following:
 * Turn off all your other free params and free frames
 * Set the checkerboard free frame to all 6DOF free
 * Run the calibration, and look at the output in the calibration.yaml for the checkerboard

guilhermelawless, 2018-06-07:
> What camera are you using?
Orbbec Astra S

> Are you sure that the checkerboard scale is correct?
If you mean the size of each square, yes sure to 0.5mm

-----------------

Here is my report from using the new branch and following your suggestion to only optimize the checkerboard frame:

- With initial offsets set to 0, results:
```
Ceres Solver Report: Iterations: 11, Initial cost: 1.194906e+00, Final cost: 1.140434e-01, Termination: CONVERGENCE
Parameter Offsets:
checkerboard_x: 0.0994882
checkerboard_y: 0.039023
checkerboard_z: 0.0170258
checkerboard_a: -0.223955
checkerboard_b: 2.19547
checkerboard_c: -2.10509
```

- Setting the initial values to the obtained values, results:
```
Ceres Solver Report: Iterations: 7, Initial cost: 2.550563e+00, Final cost: 1.140434e-01, Termination: CONVERGENCE
Parameter Offsets:
checkerboard_x: 0.0994882
checkerboard_y: 0.039023
checkerboard_z: 0.0170258
checkerboard_a: -0.223955
checkerboard_b: 2.19547
checkerboard_c: -2.10509
```

The higher initial cost is a bit strange, as well as the high values in pitch and yaw. This is our minified package: [mbot_calibration.zip](https://github.com/mikeferguson/robot_calibration/files/2079980/mbot_calibration.zip)

After building, checkerboard validation is done using
```
roslaunch mbot_calibration validate_checkerboard.launch bag:=`rospack find mbot_calibration`/bag_calibration/calibration_diag_26_poses.bag
```
And the initial offsets are changed config/mbot_check_checkerboard.yaml. I took out the camera-plane capture points for this, using only camera-chain.

Thanks for your help!

mikeferguson, 2018-06-07:
First - I've not used the Orbbec Astra yet. I think you mentioned that the camera was already calibrated separately, but us the pointcloud relatively proper sized? On the older Primesense v1.09 we found that everything was like 90% of actual size (so if you pointed it at a 20cm wide box, you only saw 18cm wide box). There were two "special" parameters in their drivers, [which I added support for](https://github.com/mikeferguson/robot_calibration/blob/melodic-devel/robot_calibration/src/models.cpp#L189).

Second - 0.5mm for the checkerboard? Is that right? That seems REALLY small.

Finally -- the a/b/c values in the YAML file are not RPY -- they are axis-magnitude. I actually wrote the offset load assuming RPY (since that is much easier for people to understand). If we convert the output a/b/c to RPY we get: -1.61582, -0.0352206, -2.97195 -- so ~pi/2 on roll, and pi on yaw. This also explains the "higher" value when you initially put things back in. I just checked and when I put the actual RPY values in, the initial and final error are the same.

mikeferguson, 2018-06-07:
Ok, I'm working on a tool for visualization (I have wanted to add one for a while). Right now it's still a mess, but I was able to visualize your bagfile:

RED = reprojection through arm
BLUE = reprojection through camera
WHITE = first point of any reprojection (helps us to be sure that orientation is correct)

First, a single checkerboard reprojection (the first one in your bag). I've applied the offsets (after fixing for RPY as in my last comment):
[image]

So I would say that orientation is correct -- that's a huge start. Error is also relatively small in this first pose (sub centimeter).

Now the full set of poses:

[image]

From this I have several comments:
 * The error gets much worse in the upper poses (~2-3cm) so a proper calibration will be very useful there.
 * Most of your poses are along a "diagonal" from lower right to upper left. We don't have much data in the upper right or lower left quadrants of the workspace.
 * Similarly, there isn't much variation in angle of the wrist. Most of the lower right quadrant have similar (within 10 degrees) orientation. This means there isn't much data for the calibration to work with.

mikeferguson, 2018-06-07:
I've added the "viz" tool and the "to_rpy" tools to my checkerboard branch, here's the launch file i've been using to run it:
```
<?xml version="1"?>
<launch>

  <node pkg="robot_calibration" type="viz" name="robot_calibration_viz"
        args="$(arg bag)"
        output="screen" required="true">
    <rosparam file="$(find mbot_calibration)/config/mbot_check_checkerboard.yaml" command="load" />
  </node>
  
</launch>
```

Then you setup a MarkerArray in RVIZ (see images above). If you want to see AFTER calibration, add the yaml file location (/tmp/calibrated_SOME_DATE.yaml) to the args.

guilhermelawless, 2018-06-07:
Thanks @mikeferguson  -- for the feedback and for the tools you are providing. So, for the small things:

1. Our Astra S looks good, we compared against laser and the scaling is fine.
2. I meant that I measured the checkerboard size with 0.5mm precision, it's actually 20.5mm as you saw in the config files
3. Oh man I didn't realize the abc-rpy thing. Maybe should also be mentioned in the README ? Thanks for the lesson!

And for bigger things:

- Today I was adding offset handling to our drivers (dynamixel). Took me a while to realize I should **add** the values when getting motor state and **subtract** when setting desired state. After applying the values mentioned before, we could finally grasp something in open loop! Tomorrow I will be using the new visualization tools (thanks!!) to check everything.

- Unfortunately our arm's reachable space is not that high, and to make things worse we don't have a joint position controller for the neck. We will be adding this so that we can capture different configurations. And I will be more careful with rotating wrist. In general, is it more important to have variation in the last or first joint?

guilhermelawless, 2018-06-11:
Current checkerboards_that_work and the backup branches are not compiling in kinetic.

`ifstream` and `istringstream` constructors using the filename need the filename as a `c_str()` before C++11. If it doesn't break melodic, that could be a fix. It's only [here](https://github.com/mikeferguson/robot_calibration/blob/checkerboards_that_work/robot_calibration/src/calibration_offset_parser.cpp#L164) and [here](https://github.com/mikeferguson/robot_calibration/blob/checkerboards_that_work/robot_calibration/src/calibration_offset_parser.cpp#L167)

mikeferguson, 2018-06-11:
Looks like that compiles fine in Melodic -- I've added a commit with that fix to the branch.

mikeferguson, 2018-06-11:
Other than the compilation issue, is that branch working for you now? Have you been able to calibrate the robot well?

closed 2018-07-07 by mikeferguson

mikeferguson, 2018-07-07:
I merged my branch since it is clearly better than what we had -- if you encounter additional bugs, please open a new ticket for them.