mikeferguson/robot_calibration #29: Kinetic: ceres-solver broken in 16.04

mikeferguson, 2016-08-23:
https://bugs.launchpad.net/ubuntu/+source/ceres-solver/+bug/1595692

k-okada, 2017-10-21:
@mikeferguson It seems ceres-solver is available on package.ros.org, so now it's ok to release fetch_gazebo in to kinetic.

It seems the all rostest has been passed
```
k-okada@p51s:~/catkin_ws/ws_fetch/build/robot_calibration$ make run_tests
Scanning dependencies of target clean_test_results_robot_calibration
Removing test result files from '/home/k-okada/catkin_ws/ws_fetch/build/robot_calibration/test_results/robot_calibration'
- removing '/home/k-okada/catkin_ws/ws_fetch/build/robot_calibration/test_results/robot_calibration/rosunit-optimization_param_tests.xml'
- removing '/home/k-okada/catkin_ws/ws_fetch/build/robot_calibration/test_results/robot_calibration/MISSING-gtest-calibration_offset_parser_tests.xml'
- removing '/home/k-okada/catkin_ws/ws_fetch/build/robot_calibration/test_results/robot_calibration/rostest-test_chain_manager_tests.xml'
- removing '/home/k-okada/catkin_ws/ws_fetch/build/robot_calibration/test_results/robot_calibration/MISSING-gtest-rotation_tests.xml'
- removing '/home/k-okada/catkin_ws/ws_fetch/build/robot_calibration/test_results/robot_calibration/rostest-test_optimization_param_tests.xml'
- removing '/home/k-okada/catkin_ws/ws_fetch/build/robot_calibration/test_results/robot_calibration/rosunit-chain_manager_tests.xml'
- removing '/home/k-okada/catkin_ws/ws_fetch/build/robot_calibration/test_results/robot_calibration/rosunit-error_block_tests.xml'
- removing '/home/k-okada/catkin_ws/ws_fetch/build/robot_calibration/test_results/robot_calibration/rostest-test_error_block_tests.xml'
[  0%] Built target clean_test_results_robot_calibration
[  8%] Built target gtest
[ 50%] Built target robot_calibration
[ 58%] Built target chain_manager_tests
[ 66%] Built target gtest_main
[ 75%] Built target rotation_tests
[ 83%] Built target optimization_param_tests
[ 91%] Built target calibration_offset_parser_tests
[100%] Built target error_block_tests
[100%] Built target tests
Scanning dependencies of target _run_tests_robot_calibration_rostest_test_error_block_tests.test
-- run_tests.py: execute commands
  /opt/ros/kinetic/share/rostest/cmake/../../../bin/rostest --pkgdir=/home/k-okada/catkin_ws/ws_fetch/src/robot_calibration/robot_calibration --package=robot_calibration --results-filename test_error_block_tests.xml --results-base-dir /home/k-okada/catkin_ws/ws_fetch/build/robot_calibration/test_results /home/k-okada/catkin_ws/ws_fetch/src/robot_calibration/robot_calibration/test/error_block_tests.test 
... logging to /home/k-okada/.ros/log/rostest-p51s-3549.log
[ROSUNIT] Outputting test results to /home/k-okada/catkin_ws/ws_fetch/build/robot_calibration/test_results/robot_calibration/rostest-test_error_block_tests.xml
testerror_block_tests ... ok

[ROSTEST]-----------------------------------------------------------------------

[robot_calibration.rosunit-error_block_tests/error_blocks_maxwell][passed]

SUMMARY
 * RESULT: SUCCESS
 * TESTS: 1
 * ERRORS: 0
 * FAILURES: 0

rostest log file is in /home/k-okada/.ros/log/rostest-p51s-3549.log
-- run_tests.py: verify result "/home/k-okada/catkin_ws/ws_fetch/build/robot_calibration/test_results/robot_calibration/rostest-test_error_block_tests.xml"
[100%] Built target _run_tests_robot_calibration_rostest_test_error_block_tests.test
Scanning dependencies of target _run_tests_robot_calibration_rostest_test_chain_manager_tests.test
-- run_tests.py: execute commands
  /opt/ros/kinetic/share/rostest/cmake/../../../bin/rostest --pkgdir=/home/k-okada/catkin_ws/ws_fetch/src/robot_calibration/robot_calibration --package=robot_calibration --results-filename test_chain_manager_tests.xml --results-base-dir /home/k-okada/catkin_ws/ws_fetch/build/robot_calibration/test_results /home/k-okada/catkin_ws/ws_fetch/src/robot_calibration/robot_calibration/test/chain_manager_tests.test 
... logging to /home/k-okada/.ros/log/rostest-p51s-3638.log
[ROSUNIT] Outputting test results to /home/k-okada/catkin_ws/ws_fetch/build/robot_calibration/test_results/robot_calibration/rostest-test_chain_manager_tests.xml
testchain_manager_tests ... ok

[ROSTEST]-----------------------------------------------------------------------

[robot_calibration.rosunit-chain_manager_tests/test_rosparam_loading][passed]

SUMMARY
 * RESULT: SUCCESS
 * TESTS: 1
 * ERRORS: 0
 * FAILURES: 0

rostest log file is in /home/k-okada/.ros/log/rostest-p51s-3638.log
-- run_tests.py: verify result "/home/k-okada/catkin_ws/ws_fetch/build/robot_calibration/test_results/robot_calibration/rostest-test_chain_manager_tests.xml"
[100%] Built target _run_tests_robot_calibration_rostest_test_chain_manager_tests.test
Scanning dependencies of target _run_tests_robot_calibration_rostest_test_optimization_param_tests.test
-- run_tests.py: execute commands
  /opt/ros/kinetic/share/rostest/cmake/../../../bin/rostest --pkgdir=/home/k-okada/catkin_ws/ws_fetch/src/robot_calibration/robot_calibration --package=robot_calibration --results-filename test_optimization_param_tests.xml --results-base-dir /home/k-okada/catkin_ws/ws_fetch/build/robot_calibration/test_results /home/k-okada/catkin_ws/ws_fetch/src/robot_calibration/robot_calibration/test/optimization_param_tests.test 
... logging to /home/k-okada/.ros/log/rostest-p51s-3814.log
[ROSUNIT] Outputting test results to /home/k-okada/catkin_ws/ws_fetch/build/robot_calibration/test_results/robot_calibration/rostest-test_optimization_param_tests.xml
[... 81 lines cut ...]

If we'd like to remove warnings, we need to fix.
```
[... cut ...]
-    Ceres
+    CERES
     orocos_kdl
   LIBRARIES
     robot_calibration
@@ -85,7 +85,6 @@ target_link_libraries(robot_calibration ${Boost_LIBRARIES}
                                         ${CERES_LIBRARIES}
                                         ${tinyxml_LIBRARIES}
                                         ${orocos_kdl_LIBRARIES})
-add_dependencies(robot_calibration robot_calibration_msgs_gencpp)
 
 add_executable(calibrate src/calibrate.cpp)
 target_link_libraries(calibrate robot_calibration
```

mikeferguson, 2018-02-13:
https://github.com/ros/rosdistro/pull/16966

closed 2018-02-13 by mikeferguson