mikeferguson/robot_calibration #30: catkin_make error

james2254, 2016-12-26:
when I catkin_make , there is an error:
`make[2]: *** 没有规则可以创建“/home/james/blockly_ws/devel/lib/librobot_calibration.so”需要的目标“/usr/lib/x86_64-linux-gnu/libspqr.so”。 停止。
make[1]: *** [robot_calibration-indigo-devel/robot_calibration/CMakeFiles/robot_calibration.dir/all] 错误 2
make: *** [all] 错误 2
Invoking "make -j4 -l4" failed
`
I just have a libspqr.so.1.3.1 in /usr/lib/x86_64-linux-gnu。

cangjiaxaun, 2016-12-28:
Well it looks like you are missing libspqr.so file.

mikeferguson, 2016-12-28:
You need to install libsuitesparse-dev. (this is actually due to upstream packaging issue in ceres-solver https://github.com/mikeferguson/ceres-solver-gbp/issues/1)

closed 2016-12-28 by mikeferguson