mikeferguson/robot_calibration #31: Using opencv calibration function to return checkerboard finder's depth information

cangjiaxaun, 2016-12-29:
Hi, I find that in the checkerboard_finder.cpp, the position of the corner seems returned by the depth information returned by the depth camera. This method, however, relay's on accurately calibrate the depth of the camera. Plus, this method also requires the checkerboard to be big enough for the camera to see them and meanwhile in the working range of the IR camera. 

So I was thinking could we use the pose of the checkerboard returned from OpenCV's calibration program instead? Have you consider this method?

mikeferguson, 2016-12-29:
@cangjiaxaun it is certainly possible to use just the 2d pixel data of where the checkerboard is located, and this is actually what the PR2 calibration used. The issue you run into is needing many more samples to properly calibrate (especially throughout different ranges). PR2 also required some "large checkerboards" at a long distance (~3-5m from the camera) for the calibration to have any chance at converging properly.

I'm not as sure about the pose, since the pose will be determined based on the camera parameters. If you reduce the entire estimation to just the pose and camera parameters, I'm not entirely sure what the implications are (since you're certainly storing less information, and it is not clear how much of the actual error information you're losing).

cangjiaxaun, 2017-01-08:
@mikeferguson Hi, mike. I'm trying your method, but I find calibrate the depth camera is not easy. I'm currently using xtion pro live. I tried the program here http://wiki.ros.org/openni_launch/Tutorials/IntrinsicCalibration. But I still get error after calibration. I also tried other program, but most of them can't execute. I wonder what camera, and what calibration program  are you use.

closed 2017-05-19 by mikeferguson