mikeferguson/robot_calibration #49: Does rviz display calibration offsets?

guilhermelawless, 2018-06-05:
Hello,

I'm sorry in advance as this does not concern directly the calibration procedure, but more the visualization of results; however, I hope it can help others as well.

We are facing some headaches when calibrating. Currently using a combination of checkerboard and ground plane (now just plane) finders. Trying to optimize similarly to what is seen in https://github.com/fetchrobotics/fetch_ros/blob/indigo-devel/fetch_calibration/config/calibrate_with_ground.yaml , but we took care of the camera calibration before.

The results of our calibration are:

> Ceres Solver Report: Iterations: 34, Initial cost: 4.190723e+00, Final cost: 4.540892e-01, Termination: CONVERGENCE
Parameter Offsets:
left_arm_joint0: -0.515459
left_arm_joint1: 0.0898604
left_arm_joint2: 0.167496
left_arm_joint3: 0.461221
left_arm_joint4: 0.166929
left_arm_joint5: -0.381634
left_arm_joint6: -2.05005
head_link_to_head_camera_link: 0.329103

We have tried to install the URDF and see no changes in rviz. All changes are made in terms of `<calibration>` tags with `rising` option. I have not successfuly found information on how this relates to `robot_state_publisher`, for instance.

On the other hand, I have tried to set some free frames (which are defined by the joints), and these values are updated in the URDF and also seen in rviz after installation.

Did you guys ever face this problem? Any tips?

Thanks in advance

mikeferguson, 2018-06-05:
The "calibration" tags are not handled by RVIZ nor robot_state_publisher -- it really has to be handled at a lower level since commands/controls need to account for these offsets. They are typically read by the drivers and added to the raw joint value before publishing the joint_states or sending the joint values into the controllers.

The "rising" part is really a legacy of the PR2 (due to slack in belts, the rising vs falling calibration might be different) -- but it's the "official" tag in URDF.

guilhermelawless, 2018-06-05:
I see, thank you @mikeferguson  this helped us a lot. Closing as answered.

mikeferguson, 2018-06-05:
I changed the title and re-opened this, as I'd like to move some of my answer into the readme. This will remind me to do so sometime in the next week or so.

guilhermelawless, 2018-06-07:
@mikeferguson  although these values are "offsets", they should actually be added and not subtracted from joint values, right? This is also something that took some time for me to notice.

mikeferguson, 2018-06-07:
Yes, they are added -- [since that is how the offset is used in the model](https://github.com/mikeferguson/robot_calibration/blob/2ff79fa4f486d1b3c8ced156243029f8fc236423/robot_calibration/src/models.cpp#L131). I'll make the README a bit more specific.

commit da80068e (2018-06-06): add some notes on URDF usage (closes #49)
commit 8a311aab (2018-06-08): add some notes on URDF usage (closes #49)
closed 2018-06-08 by mikeferguson