mikeferguson/robot_calibration #54: Add a "pose capture" mode

guilhermelawless, 2018-06-11:
Using a [script](https://github.com/mikeferguson/robot_calibration/blob/master/robot_calibration/scripts/capture_poses) to capture poses when a whole pipeline is built to perceive and calibrate seems out of place.

Not only that, but at least in my case, i needed to edit the script to add a line with `self.last_state_.features = ['checkerboard_finder', 'ground_plane_finder']` or whatever my captured features will be, or else it wouldn't work during perception. This might be a bug in itself.

Nontheless, my suggestion is a **--pose-capture BAG** mode which would be very similar to **manual** mode but would not enable perception (since the person is likely moving the arm by itself and is in the FOV of the camera). In this mode a bag would be saved with `/joint_states`.

This mode would enable a feature which I describe in another issue and will cross-ref to this one.

mikeferguson, 2018-06-11:
Yup, that appears to be completely out of date now that you point it out. Once upon a time when we didn't have many finders, we didn't specify finders at all in the bag, since we ran all feature finders on all poses (that obviously works terribly when you want to do ground calibration).

I do like the idea of capturing within the main node (or a setup wizard). I've got something similar called on here: https://github.com/mikeferguson/robot_calibration/issues/51 (but that is a longer term project).

As for not enabling perception, I'd suggest that be a setting too. I presume you aren't applying gravity compensation on your Dynamixel arm, correct? Many robots use gravity compensation such that once the arm is placed in a pose, it will remain there. For instance, when using manual mode on the Fetch, I would move the arm to a position, step away to hit enter on the keyboard, and then step back.

mikeferguson, 2018-06-11:
I'll also note, that script was typically run by itself, we didn't usually have calibration running beside it, we would just look at the camera view through image_view or rviz to make sure the target was visible.

guilhermelawless, 2018-06-12:
We don't have compensation which we solve in this case by having two people, one to move and hold the arm and the other to check if checkerboard is visible on rviz and press to save the pose.

Glad you like the idea, it made sense to me!

mikeferguson, 2024-11-02:
Now tracking in https://github.com/mikeferguson/robot_calibration/issues/180

closed 2024-11-02 by mikeferguson