mikeferguson/robot_calibration #58: LED finder not work for any other number of led than 4

saurabhbansal90, 2018-06-23:
Hello Mark,

I am using `robot_calibration` package to calibrate my robot. And its a very helpful.
I am using only 1 led for calibration. So I defined only 1 led in the .yaml file.

But it was not working. So I got the issue and would like to suggest a solution for it as below:
- In `led_finder.cpp` file ([link](https://github.com/mikeferguson/robot_calibration/blob/master/robot_calibration/src/finders/led_finder.cpp)) at `line number 188`, the code is as below:

> code_idx = (code_idx + 1) % 8;

So this can be replaced with 

> code_idx = (code_idx + 1) % codes_.size();

Can you please have a look over it and change in main code.

Regards,
Saurabh

mikeferguson, 2018-06-23:
That seems right. Please open a pull request if you have time.

Separate note, a single led is usually challenging to get good calibration (we only had one on PR2) and so it is really hard to determine the angles of the wrist (especially if the led is too close to the axis).

Lastly, would love to know more about your robot / use case if you are willing to share.

saurabhbansal90, 2018-06-24:
Hello Mike,

I would love to create a pull request for this, but it is showing, I have no permission for it.
This is my first pull, can you please let me know if there is some alternative.

We are developing a robot arm for cotton harvesting and the aim is to just reach upto the cotton-bud. There is no gripper attached to end-effector. So I thought only 1 led might be enough for calibration.

Please let me know your inputs on this.

Saurabh

mikeferguson, 2018-06-24:
@saurabhbansal90 -- as for permission, that's typical. As with most projects on Github, you can fork the project, make your changes in your fork and open a pull request from that into my project. See https://help.github.com/articles/creating-a-pull-request-from-a-fork/ for more details.

On the calibration, If you aren't calibrating the joint angle offsets (just doing camera to end effector) and the orientation isn't important, a single led might be OK. It has been my experience that it is very difficult to get enough information about the kinematics from a single LED when any parameter of the arm kinematics itself is a free parameter.

mikeferguson, 2018-06-25:
#59

closed 2018-06-25 by mikeferguson