mikeferguson/robot_calibration #61: Problem in waitToSettle() function in manual mode

saurabhbansal90, 2018-07-04:
Hello Mike,

There was 1 more issue in the function waitToSettle(), if we use the manual mode to move the arm.

1. At line 269 in function waitToSettle() in [file](https://github.com/mikeferguson/robot_calibration/blob/master/robot_calibration/src/chain_manager.cpp), ros::spinOnce() is being called. Because of this spin call, the callback will be called and joint state values will be updated.
2. It will be working without any problem, if arm is moved as per poses in calibration_poses.bag file, as after the movement, it will take sometime to settle the arm.
3. So at first iteration in while(true) loop at line 236, the variable settle will be false and this loop will not break at line 266.
4. But if we are moving the arm manually like mine case, I am first moving the arm to some desired position and then pressing enter to capture the pose.
5. In this case, the arm is already settled, when it comes in waitToSettle() function.
6. So loop will break at line 266, and ros::spinOnce() will never called.
7. So we will not get the joint states.

So here I would like to propose a solution:
1. We can place ros::spinOnce() before line 266.
2. Or we can place ros::spinOnce() at the beginning or at last only, so that inspite of whether arm has been settled or not, ros::spinOnce() should be called.

Please let me know if you agree or find any issue with it.

Thanks,
Saurabh

mikeferguson, 2018-07-04:
Yup, that seems about right. Moving ros:spinOnce() probalby fixes it most of the time -- but there is one other edge case -- that no message arrives in time for your spinOnce() and so the message is stale. There is a comment in getState() -- that we should return whether state is valid (right now it always returns True).

I think #62 is a complete fix -- could you give it a try?

mikeferguson, 2018-07-04:
Also -- the other edge case I note could also (possibly) hit users in non-manual mode (although it is fairly unlikely)

PR #62 (2018-07-05): make sure we get valid joint_states
* invalid old state, wait for new message
* fixes #61
commit fe97d03f (2018-07-05): make sure we get valid joint_states

* invalid old state, wait for new message
* fixes #61
closed 2018-07-05 by mikeferguson