mikeferguson/robot_calibration #66: Use of Joint States in Led Finder

saurabhbansal90, 2019-01-28:
Hello Mark,
I have seen in the code that joint states are being captured along with sensors observations.
Can you please let me know if there is any use joint states, when we use led-finder for calibration.

Regards,
Saurabh

mikeferguson, 2019-02-04:
@saurabhbansal90 Apologies on delayed response, just getting back from out of office.

The joint states are always used -- we need them to know the configuration of the kinematic chain (where the arm/head is for a given set of sensor observations).

Are you using a single LED? I've never had good results with a single LED -- the Fetch/UBR both had a pattern of 4 leds so that rotation of the wrist joint can be determined. The PR2 actually had a single LED and we never used it because it did not provide enough data (further, it was on-axis with the wrist roll joint, so it gave even less rotation data than if it had been significantly off-axis. we always used checkerboards with the PR2 calibration). If you are trying to use a single LED, I'd suggest you need MANY samples (it appears you have 14 -- I would suggest a single LED might need 100-200 samples to have any hope of working).

With only a single LED and only 14 samples, I could definitely see things getting overfit to the data (which could result in really wonky URDF changes). It's possible to try and cancel this out with the use of an "outrageous error block" -- which basically adds an error term for how much a given joint offset changes, but the better approach would be to add more samples, and change to a checkerboard or multiple LED pattern. (For reference, with the Fetch robot, which has 9 joints under calibration (7 arm Dofs + 2 head dofs) along with a 6-dof pose of the camera link, we needed about 50-70 well-distributed poses to get good results, maybe 10-15 less for a checkerboard).

saurabhbansal90, 2019-02-07:
Hello Mike,

Thanks for the response, actually I was there at our field visit. So was not able to reply back, sorry for that.
Actually we were not using the LED's, but a similar program using a marker. The reason was that, the LED finder program was not able to detect the depth of LED point, as we were using a Stereo Camera, instead of IR camera.
So we made a marker, a unique pattern and tied that with the tip of our arm, and was detecting its xyz location and performed calibration.
I was using 14-15 points and 3 joints for calibration, and by god'd grace it was all working fine.

After that, there were few changes in the mechanical design and so in URDF.
After that something gone bad with calibration, and till now I am not able to make it work again.
To add more points and to be able to calibrate in open field also, I used Aruco pattern in the marker.

So now I have 4 points in one observation. And point number are same as 15, and number of joints to be calibrated is also same i.e. 3 joints.
But I am getting very vague values. I tried to print the value of INITIAL COST also, that seems to be fine.
But still not able to get good calibrated URDF.

I am not sure if its a problem with local minima. Can you please suggest some methods to  debug and to find the problem with my configuration or somewhere.
Or is there any method to print the value of cost function so that I  can check if its stuck in a local minima.

Regards,
Saurabh

closed 2022-01-14 by mikeferguson