mikeferguson/robot_calibration #77: ASSERTION FAILED during chain_manager_tests

jbeck28, 2020-03-06:
I'm running the chain_manager_tests.py after manually loading the yaml files in the tests (after encountering the same issue with my own yaml files I created), and I get: 

```
[FATAL] [1583526733.187227715]: ASSERTION FAILED
	file = /home/joshua/catkin_ws/src/robot_calibration/robot_calibration/src/chain_manager.cpp
	line = 37
	cond = chains.getType() == XmlRpc::XmlRpcValue::TypeArray

```

Any guidance would be helpful, 
Thanks

mikeferguson, 2020-03-06:
Interesting, it seems to not be failing on standard 18.04/melodic - can you tell me for your setup:

 * what operating system?
 * what ROS distro (and how installed)

jbeck28, 2020-03-08:
Thanks for the quick reply, and I apologize for not including this information in my original post. I'm running Ubuntu 18, with ROS melodic. I installed ROS from the repos.

mikeferguson, 2020-03-09:
This is really strange - I can't reproduce locally on my 18.04/melodic machine (I even did an apt-get upgrade to make sure I had all the latest debs, still not happening). The buildfarm is also passing tests: https://github.com/mikeferguson/robot_calibration/pull/78. I'm not sure what's going on.

jbeck28, 2020-03-09:
I think I figured it out. It came down to user error. I hadn't run the chain_manager_tests.test yet, and so the parameters were not on the server yet. To be clear, I was trying to run the built in the tests with built in yamls files because they didn't run when I made my own yaml files. 

Thanks again for your help, sorry for the false alarm!

jbeck28, 2020-03-09:
Pretty sure this comes down to user error. I'll likely open a separate issue with some questions on how to properly set this up.

closed 2020-03-09 by jbeck28