mikeferguson/robot_calibration #79: openCV adaptive threshhold error

jbeck28, 2020-03-11:
I'm running Ubuntu 18.04, ROS Melodic, installed from the repos. I've found that when running `calibrate`

the following error pops up when this line of code in `checkerboard_finder.cpp` gets run:

```
  int found = cv::findChessboardCorners(bridge->image, checkerboard_size,
                                        points, CV_CALIB_CB_ADAPTIVE_THRESH);
```

```
OpenCV Error: Assertion failed (blockSize % 2 == 1 && blockSize > 1) in adaptiveThreshold, file /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/imgproc/src/thresh.cpp, line 1447
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/imgproc/src/thresh.cpp:1447: error: (-215) blockSize % 2 == 1 && blockSize > 1 in function adaptiveThreshold

```

mikeferguson, 2020-03-12:
I took a quick look into this -- but I'm not sure what is going on (it is several layers deep in openCV). It'll take me a couple days to get a chance to run calibration on a robot - I don't recall for sure, but I'm pretty certain I've tested the checkerboards under 18.04.

Is it possible that the config you are using for your robot is posted somewhere publicly? Most issues tend to be a configuration issue (although, I'm not sure what issue could cause *this*)

jbeck28, 2020-03-12:
By the end of today I should be able to get my configs together and put them on the internet. You're my gut says you're likely correct since I didn't (don't) have a full understanding of what I'm doing. 

Thanks again for your support in this.

jbeck28, 2020-03-12:
https://github.com/jbeck28/abb_robot_calibration

Here are the config files.

jbeck28, 2020-04-01:
I realized a few weeks ago (before I stopped going to work due to social distancing measures here in NY) that my checkerboard frame is Z up, rather than (if memory serves) your gripper frame being Y up..? I'm not sure if this would cause the error, or simply give Ceres a hard time, but I thought I'd throw this out there.

jbeck28, 2020-05-27:
Just another update, I tried using this package when gotten from apt, and when compiled from source on a new computer, with no luck. I also compiled against the latest version of openCV, 3.4.10, rather than the typical 3.2. Same error: 

`terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(3.4.10) /home/spare-desktop/opencv-3.4.10/modules/imgproc/src/thresh.cpp:1665: error: (-215:Assertion failed) blockSize % 2 == 1 && blockSize > 1 in function 'adaptiveThreshold'

================================================================================REQUIRED process [capture-1] has died!
process has died [pid 9166, exit code -6, cmd /home/spare-desktop/catkin_ws/devel/lib/robot_calibration/calibrate --manual __name:=capture __log:=/home/spare-desktop/.ros/log/3d6041a8-a04b-11ea-9319-0013eff217a1/capture-1.log].
log file: /home/spare-desktop/.ros/log/3d6041a8-a04b-11ea-9319-0013eff217a1/capture-1*.log
Initiating shutdown!
================================================================================`

If you have any ideas on how you, @mikeferguson would troubleshoot this, perhaps I could help if you can point me in a direction?

jbeck28, 2020-05-28:
I figured it out. For anyone using an Intel RealSense, or any camera for that matter, make sure that your pointcloud2 messages are organized. Unorganized pointclouds cause this issue.

mikeferguson, 2020-05-28:
Interesting - I'm gonna leave this ticket open then, and add a check for organized cloud so people don't run into this.

commit 6b38e1fa (2020-06-05): only accept organized clouds, fixes #79 (#90)
closed 2020-06-05 by mikeferguson