mikeferguson/robot_calibration #94: What is the "base frame" where poses are defined?

AndyZe, 2021-08-06:
First time using this package. It looks really flexible!

I'm doing a hand-eye calibration. Checkerboard is stationary in the world, camera is mounted on the robot end effector.

Question - what frame is the "free_frames_initial_values" relative to? My guess is, it's relative to the base link of the kinematic chain.

Happy to make a PR if you give me a little guidance.

mikeferguson, 2021-08-06:
The "base_frame" is basically the root of your kinematic tree during the calibration (it will be the coordinate system used when calibrating). For instance, with the Fetch robot:

 *  The root of the kinematic tree could be base_link (since that is the bottom of the robot), [but we actually end up making it torso_lift_link for generic calibration](https://github.com/fetchrobotics/fetch_ros/blob/melodic-devel/fetch_calibration/config/calibrate.yaml) (since we don't move that joint).
 * Later on, support was added for calibrating the sensor to the ground plane (as viewed by the sensor) - and for that type of calibration we had to update the [base_frame to base_link](https://github.com/fetchrobotics/fetch_ros/blob/melodic-devel/fetch_calibration/config/calibrate_with_ground.yaml), since the ground plane calibration assumes that "z=0" is how you define the ground.

For a fixed arm versus a stationary camera, I assume you have a base_link for the arm in your URDF, and then there is a fixed joint in your URDF that tells where the camera is located relative to that base_link. Basically, whatever is the root of your URDF should be the base_frame for your scenario.

mikeferguson, 2021-08-06:
For the free_frames_initial_values - these are basically offsets for the particular frame. With your checkerboard, it is typically connected to something like the wrist_roll_link. If you did not set free_frames_initial_values for the checkerboard frame, it would assume that your virtual "checkerboard" frame has an Identity matrix transformation from the "wrist_roll_link". Usually this is not correct - and so it cause the calibration to take a really long time to converge (especially for an incorrect rotation). The initial values are basically added to that identity transformation to specify what the initial offset would be. (these initial values can also be used to correct a known kinematic error in the URDF - but that's less common)

mikeferguson, 2021-08-06:
And PRs for documentation would be super awesome!

AndyZe, 2021-08-09:
Thanks. That makes a lot of sense. I guess I missed the `base_link` parameter.

mikeferguson, 2021-08-10:
Thanks for the PR to update documentation!

closed 2021-08-10 by mikeferguson