mikeferguson/robot_calibration #95: error_block type "camera3d_to_arm" is not documented

AndyZe, 2021-08-06:
In the Readme, I've noticed that **camera3d_to_arm** type is not documented.

Happy to make a PR to improve the documentation if you give me some guidance. My guess is:

`camera3d_to_arm - This is a specialization of the chain3d_to_chain3d type. Useful, for example, when the camera is fixed in the world and the robot moves the chekerboard.`

mikeferguson, 2021-08-06:
So camera3d_to_arm just adds the option for camera intrinsics calibration - it makes no assumption about whether the camera is fixed or moving (as far as I remember). Basically, a chain just does the transformation for each joint, but the camera version also adds pinhole parameters (cx, cy, fx, fy) when doing the projection of the points. Depending on your setup, this doesn't always work (for instance, some drivers don't actually use the camera model/parameters for the 3d projection, but the Primesense/Astra ones do for instance).

mikeferguson, 2021-08-10:
Documentation updated

mikeferguson, 2021-08-10:
And right after I merged that I noticed that the naming didn't match - the reason camera3d_to_arm isn't documented is that we renamed it chain3d_to_chain3d starting with Melodic - you can see how the Fetch config got updated here: https://github.com/fetchrobotics/fetch_ros/commit/eb7dd42534ac19b4b71b0cc75f93dee1de8b902c#diff-94b793231b80eb1ddedc51d3af8c75bce2f94da77c478b32fe94e99e953d632b - but there is still some remnants in this codebase (mainly in the tests). My comments about projection matrices should be applied to the difference between the `camera3d` and `chain` models rather than the error blocks (they just leverage the models).

I've created https://github.com/mikeferguson/robot_calibration/pull/99 to fix these references and update the documentation.

mikeferguson, 2021-08-10:
#99 merged

closed 2021-08-10 by mikeferguson