mikeferguson/robot_calibration #139: Does it matter if the checkerboard is rotated 180* in some samples?

AndyZe, 2022-07-14:
After a large amount of effort, we are finally getting good hand/eye, eye/eye, and kinematic calibrations from this package  :+1:

But I've been wondering this:

We have a checkerboard target bolted to the end effector. Does it matter it we accidentally flip the checkerboad 180* while collecting samples? It seems like it would cause a ton of error since the checkerboard looks the same if rotated 180* and the pose of the checkerboard is what matters (if I understand correctly).

Here's what I mean. We do kinematics calibration something like this:
[image]

But while moving the robot to collect samples, we accidentally rotate the checkerboard like this. The checkboard pose probably still looks the same to the camera.
[image]

mikeferguson, 2022-07-14:
Yeah - I've run into this before when I just grabbed any old checkerboard - the trick is you need to use a checkerboard that isn't symmetric. Based on your image, if you have a checkerboard with one less row in either of the dimensions then the checkerboard finder will always find the correct orientation

AndyZe, 2022-07-14:
Duh - thanks

closed 2022-07-14 by AndyZe