mikeferguson/robot_calibration #168: Is it possible to set link lengths as a free parameter?

BenBlumer, 2024-08-16:
Thanks for the package, Mike!

I'm using ros2-Jazzy. I've been experimenting with trying to get link lengths to calibrate. I have a URDF containing a single link robot with base_link, rotating_link (**1** meter long), end_of_rotating_link, fixed_end_of_end_effector_joint, base_to_rotating_joint and a camera_link (fixed relative to base_link). 

I have a script to drive that link to rotate in a circle and get the X,Y,Z position of the end_of_end_effector from TF2. I re-publish this info as CalibrationData messages. I record a rosbag with the calibration data. 

I then modify the rosbag so that the robot_description now has a link **1.1** m long. 

I feed that rosbag into robot_calibration hoping for it to find the discrepancy in the link_length. 

```
robot_calibration:
  ros__parameters:
    verbose: true
    base_link: base_link
    calibration_steps:
    - single_calibration_step
    single_calibration_step:
      models:
      - arm
      - camera
      arm:
        type: chain3d
        frame: end_of_rotating_link
      camera:
        type: chain3d
        frame: camera_link
      free_params:
        - rotating_link
      error_blocks:
       - hand_eye
      hand_eye:
        type: chain3d_to_chain3d
        model_a: camera
        model_b: arm
```

However, the output I'm getting doesn't seem right.

```
Solver output:
iter      cost      cost_change  |gradient|   |step|    tr_ratio  tr_radius  ls_iter  iter_time  total_time
   0  6.190203e+00    0.00e+00    0.00e+00   0.00e+00   0.00e+00  1.00e+04        0    1.69e-03    1.79e-03

Ceres Solver Report: Iterations: 1, Initial cost: 6.190203e+00, Final cost: 6.190203e+00, Termination: CONVERGENCE
Parameter Offsets:
rotating_link: -1.49012e-08
```

So, am I doing something wrong? Or does the feature not exist yet?

mikeferguson, 2024-08-16:
It is possible to calibrate link lengths:

 * In this case you would not have a "free_param" but rather a "free_frames" - and then you would specify which axis is to be calibrated. You can see an example here where I calibrate the full 6 DOF pose of the camera: https://github.com/mikeferguson/ubr_reloaded/blob/ros2/ubr1_calibration/config/calibrate.yaml#L37-L47
 * One thing to keep in mind: in practice, I've found calibrating link lengths requires vastly more samples than calibrating joint offsets

mikeferguson, 2024-08-20:
Added notes on calibrating link length to README: https://github.com/mikeferguson/robot_calibration/commit/4220f4519720b58af3951c1e6c81fa8335142823 - closing this as solved - but if you have additional issues, please reopen.

closed 2024-08-20 by mikeferguson

BenBlumer, 2024-08-21:
Hey Mike,

Thanks for your help with this! It took me a beat, but I got something to work using the example you linked to. 

I'd love to contribute it -- I think it could be an accessible entry point to folks newer to robot_calibration. It's commented in a way that should help people understand what's going on. Want me to?