mikeferguson/robot_calibration #170: symbol lookup error on ROS2 jazzy

Tacha-S, 2024-09-12:
When I run a package installed with apt, the following error occurs.

```bash
[calibrate-1] /opt/ros/jazzy/lib/robot_calibration/calibrate: symbol lookup error: /opt/ros/jazzy/lib/librobot_calibration_feature_finders.so: undefined symbol: _ZN3tf212getTimestampIN13geometry_msgs3msg13PointStamped_ISaIvEEEEENSt6chrono10time_pointINS6_3_V212system_clockENS6_8durationIlSt5ratioILl1ELl1000000000EEEEEERKT_
```

mikeferguson, 2024-09-12:
Interesting - I haven't run into this. Which feature finders are you using?

Tacha-S, 2024-09-13:
I am using `robot_calibration::PlaneFinder` and `robot_calibration::ScanFinder`. When I installed them via apt, I encountered a symbol lookup error, but the error did not occur when I built them from source. There might be an issue with the apt package's binary.

mikeferguson, 2024-09-26:
Ok, so returning to looking at this:

 * getTimestamp() is inlined (and templated)
 * internally it calls tf2_ros::fromMsg - which is also inlined and templated
 * nm/ldd don't seem to show any issues with calibrate, librobot_calibration, or librobot_calibration_feature_finders on my AMD64 install of most recent jazzy binaries.

Are you on some different platform? Something other than AMD64 on Ubuntu 24.04?

Tacha-S, 2024-09-27:
I'm executing on AMD64 with Ubuntu 24.04, just like you.

mikeferguson, 2024-09-28:
I put a new release out for Jazzy+Rolling yesterday which includes the fixes I need to be able to run calibration on my robot - as soon as those get built I'll try out the debs and try to figure out what is going on

mikeferguson, 2024-11-08:
Have reproduced this locally:

 * Jazzy deb is missing the symbol
 * Totally up-to-date 24.04 source build has the symbol
 * Iron deb is also missing the symbol
 * Humble deb is also missing the symbol

Building with -DCMAKE_BUILD_TYPE=None (as debs are), still generates the symbols...

But building with -DCMAKE_BUILD_TYPE=RelWithDebInfo gets the missing symbols (and harks back to a similar issue I had here: https://github.com/mikeferguson/graceful_controller/issues/52)

This missing (inline) function is being used in tf2_buffer_->transform(), so we can go find where that was included (include/feature_finder.hpp) and move the geometry_msgs import ahead of that - and voila, our RelWithDebInfo build no longer has the undefined symbol (in fact, symbol is entirely gone - which is what i would expect since it is inline).

mikeferguson, 2024-11-08:
Jazzy/rolling fix should be: https://github.com/mikeferguson/robot_calibration/pull/182 (and a new release)

mikeferguson, 2024-11-12:
I just grabbed the latest Debian build of 0.9.2 for Iron from the ros2-testing - the missing symbol issue is resolved. This problem will go away fully once the next sync happens for Humble, Iron, Jazzy and Rolling distributions

closed 2024-11-12 by mikeferguson